rust-embedded/gpio-cdev #3: Multi-bit access

fpagliughi, 2018-09-04:
Hi! Do you have a plan, in the near future, to implement the access of multiple bits simultaneously? If not, do you want me to take a crack at it?  I'm going to need this within the next few weeks.

fpagliughi, 2018-09-14:
Turns out the initial need was mainly to be able to add individual input bit monitoring to an app that was already using poll() for a number of other non-GPIO file handles. This was solved pretty easily with #4 

I hadn't seen it documented, but apparently the line event handle can also be used to read the current value of the bit. So it would be great to add a `get_value(&self)` to the `LineEventHandle` implementation.

And, finally, the actual multi-bit access might be implemented with a `Lines` (plural) struct. It would mimic the `Line` struct, but take a slice/array/Vec of offsets and values for each call where the `Line` struct only takes or returns a single value.

posborne, 2018-09-26:
Down the road adding support for handles/events over multiple lines will definitely be desired.  It doesn't fit a lot of use cases as all the lines need to be configured the same way (this is a limitation of the kernel) but does have some nice performance benefits in some specific cases so it is something we'll want to do eventually.

Probably once #8 lands the next focus will be getting a release out and seeing about moving things to the rust-embedded org and getting feedback on ergonomics so we can get any breaking changes in early.

fpagliughi, 2018-09-28:
That sounds great. I've made some progress on the multi-line stuff, and might have something as early as next week. But I agree that we shouldn't hold up what we have already for a release. The multi-line stuff can come in later as v0.2 or beyond.

fpagliughi, 2018-12-12:
Done in PR #10.

closed 2018-12-12 by fpagliughi