rust-embedded/gpio-cdev #5: Represent line levels as a bool

fpagliughi, 2018-09-14:
I know the char driver(s) expose bits as `u8` values, but it might be easier to manipulate the values as booleans. Should we consider:
```
impl LineHandle {
    pub fn get_value(&self) -> Result<bool> { ... }
    pub fn set_value(&self, value: bool) -> Result<()> { ... }
    ...
```

posborne, 2018-09-26:
The actual value (0, non-zero) is modeling "active" state for the line which can be inverted in the kernel via the active-low setting.  I could see `is_active` and `set_active` as an API that modeled what this is in the kernel being something that make sense.  This could be in addition to the value or as a full replacement.

fpagliughi, 2018-09-30:
Whenever I wrap a C library or API in a language that has a bool type (C++, Rust, etc), I usually prefer to convert the C integer boolean (0, non-zero) to a bool to be a little more explicit about the possible values, and remove ambiguities. But it's no big deal either way.

rubberduck203, 2020-08-18:
Possibly relevant. I've been working on a reusable abstraction for logic levels.
The idea is to abstract over whether an I/O line is active high or low and only expose whether or not it is currently asserted.

https://github.com/rubberduck203/switch-hal