rust-embedded/gpio-cdev #18: How to do monitor for an event on more than one input?

andy-sdc, 2019-08-23:
How do you extend the gpioevents.rs example to handle multiple input pins?

-Andy.

posborne, 2019-08-23:
Polling using your mechanism of choice is fine -- Providing a way to conveniently integrate this lib with tokio is not in the lib today but is something that probably should be added (probably behind a feature flag).

Events on the GPIO are queued using cdev so you will not miss them unless you really are not keeping up in your userspace application.  Each event comes with a [`timestamp`](https://docs.rs/gpio-cdev/0.2.0/gpio_cdev/struct.LineEvent.html#method.timestamp) for interrupt context which is what should be used if you care about when an event really happened.

If the lines are on the same gpiochip you can read multiple using a MultiLineHandle (see https://github.com/rust-embedded/gpio-cdev/blob/master/examples/multiread.rs).  The equivalent of `events` on this type of handle is not implemented at present although I think it should be possible (though I haven't tested this).

massimiliano-mantione, 2020-05-29:
> The equivalent of `events` on this type of handle is not implemented at present although I think it should be possible (though I haven't tested this).

Could you please provide a pointer to the docs of the C API that specify how to do it?
I am really interested in this and could provide a PR.
I would test it reading the two lines of a quadrature encoder (this is the use case I have).

rubberduck203, 2020-09-23:
I don't think #35 should have closed this issue.
While it's possible to monitor multiple lines for events, it's not nearly as ergonomic as the single line API.
I would personally love to see something like the `Line::events()` and `LineEventHandle` iterator implemented for `Lines` and (a new?) `MultiLineEventHandle`.

commit 448d6fd1 (2020-08-02): Merge #35

35: RFC: Line event stream support for Tokio r=posborne a=mgottschlag

The second commit (on top of #34 for my convenience, but completely independent) adds support for asynchronous handling of line events/interrupts based on Tokio. This patch probably fixes #18 .

This probably needs some further discussion:

* I set the Rust edition to 2018 to be able to use `async`/`await` in the example.
* The API is just a light wrapper around `LineEventHandle`. I implemented `AsRef<LineEventHandle>` instead of adding a separate `get_value()` function. Do we want a function to destroy an `Async [... cut]
PR #35 (2020-08-02): RFC: Line event stream support for Tokio
The second commit (on top of #34 for my convenience, but completely independent) adds support for asynchronous handling of line events/interrupts based on Tokio. This patch probably fixes #18 .

This probably needs some further discussion:

* I set the Rust edition to 2018 to be able to use `async`/`await` in the example.
* The API is just a light wrapper around `LineEventHandle`. I implemented `AsRef<LineEventHandle>` instead of adding a separate `get_value()` function. Do we want a function to destroy an `AsyncLineEventHandle` and get the original `Li [... cut]