rust-embedded/gpio-cdev #36: Errors arn't Sync

bestouff, 2020-05-25:
Impossible to use `anyhow` with `gpio-cdev`:

```
error[E0277]: `(dyn std::error::Error + std::marker::Send + 'static)` cannot be shared between threads safely
 --> src/main.rs:9:52
  |
9 |     let mut chip = gpio_cdev::Chip::new(RELAY_CHIP)?;
  |                                                    ^ `(dyn std::error::Error + std::marker::Send + 'static)` cannot be shared between threads safely
  |
  = help: the trait `std::marker::Sync` is not implemented for `(dyn std::error::Error + std::marker::Send + 'static)`
  = note: required because of the requirements on the impl of `std::marker::Sync` for `std::ptr::Unique<(dyn std::error::Error + std::marker::Send + 'static)>`
  = note: required because it appears within the type `std::boxed::Box<(dyn std::error::Error + std::marker::Send + 'static)>`
  = note: required because it appears within the type `std::option::Option<std::boxed::Box<(dyn std::error::Error + std::marker::Send + 'static)>>`
  = note: required because it appears within the type `error_chain::State`
  = note: required because it appears within the type `gpio_cdev::errors::Error`
  = note: required because of the requirements on the impl of `std::convert::From<gpio_cdev::errors::Error>` for `anyhow::Error`
  = note: required by `std::convert::From::from`
```

nastevens, 2020-05-31:
This is a side-effect of using error-chain, which isn't really needed with the stabilization of good `std::error::Error` support. Moving off of error-chain seems reasonable to me. I'm going to retitle this to reflect the move from error-chain to `std::error::Error`

bestouff, 2020-06-01:
You should probably consider using [`thiserror`](https://docs.rs/thiserror), which is a helper for writing `std::error:Error`-derived errors and doesn't add anything to the exported interface.

fpagliughi, 2020-06-01:
I've been using `thiserror` lately and am pretty happy with it.

I can convert this project to use `thiserror` if you want. It's easy enough.

fpagliughi, 2020-06-01:
OK. Now I'm confused. `src/errors.rs` seems to have a hand-written std::errors implementation with the examples using `anyhow` from last Oct. 

Has this just not been published to crates.io?

nastevens, 2020-06-01:
> Has this just not been published to crates.io?

Looks like that is the case - `Cargo.toml` has even been updated to 0.3 but 0.2 is what's on crates.io. I'll pester @posborne and see if we can get this released.

posborne, 2020-06-01:
0.3.0 is now published: https://crates.io/crates/gpio-cdev.  Sorry for the delay and thanks for the ping @nastevens

nastevens, 2020-06-01:
Thanks Paul - closing this as anyhow is now properly supported 👍

closed 2020-06-01 by nastevens

bestouff, 2020-06-01:
Thanks everyone !

fpagliughi, 2020-06-01:
Well that was easy.