rust-embedded/gpio-cdev #38: gpio-cdev = { version = "0.3", features = ["async-tokio"] } fails

emilhoeiriis, 2020-08-20:
Create a new project `foo` with `gpio-cdev` as a dependency with the _async-tokio_ feature.
```
[dependencies]
gpio-cdev = { version = "0.3", features = ["async-tokio"] }
```
This gives us an error
```
... the package `foo` depends on `gpio-cdev`, with features: `async-tokio` but `gpio-cdev` does not have these features.
```

posborne, 2020-08-23:
@ryankurte Thanks for the bump; that reminds me that I have a branch doing some polish sitting on my drive ready to be finished up to prep the next release.  I'll get that up.

posborne, 2020-08-23:
https://github.com/rust-embedded/gpio-cdev/pull/40

posborne, 2020-08-25:
0.4.0 is now published https://crates.io/crates/gpio-cdev

closed 2020-08-25 by posborne