rust-embedded/gpio-cdev #42: GPIO electrical configuration

dbrgn, 2020-11-02:
Is there support for electrical configuration of GPIO pins (e.g. enabling pull-up resistors on the Raspberry Pi)?

If I look at https://www.kernel.org/doc/html/v5.3/driver-api/gpio/driver.html#gpio-electrical-configuration this should be possible, right? Or is that a different API? (I'm still figuring out the kernel GPIO APIs, so forgive me if I'm mixing up things :slightly_smiling_face:)

posborne, 2020-11-02:
@dbrgn Nope, this isn't part of the userspace API.  Typically, this is done via devicetree as part of the board setup for custom hardware.  For hobbyist boards which are less fixed there are DT Overlays.  Configuration of these characteristics of GPIOs are often highly IP chunk dependent as the pin config usually requires muxing, etc.

https://www.raspberrypi.org/documentation/configuration/device-tree.md

posborne, 2020-11-02:
Ok, I might have spoken a bit too soon as some new stuff landed in kernel 5.5.  I'll leave this open and get back to it once I do some more reading.

posborne, 2020-11-02:
https://microhobby.com.br/blog/2020/02/02/new-linux-kernel-5-5-new-interfaces-in-gpiolib/ has a nice summary (libgpiod focused).  There's a new V2 ABI for GPIO which we'll need to update to support (while keeping some compatability with the V1 ABI for older kernels).

sadilekivan, 2022-08-10:
Hi, I have a work project that depends on internal resistor setup. Was some work done on this or should I just fork.

eldruin, 2022-08-11:
There have been some discussion over at #68, I am not sure if some progress has been made.

JonathonReinhart, 2023-12-18:
> Was some work done on this or should I just fork.

Care to elaborate on what "*just fork*" means? I'm going to assume that you mean "*just* [click the] *fork* [button on GitHub and send a pull request]".