rust-embedded/gpio-cdev #43: How would one configure a PWM using this crate?

acburigo, 2020-11-03:
Is there an easy way to configure a PWM using this crate? If not, does anyone know a good crate for doing so?

fpagliughi, 2020-11-06:
No, this crate does not expose the Linux PWM interface, if that's what you're looking for.

acburigo, 2020-11-06:
Is this a limitation of this crate or a limitation of "GPIO Character Device"? I've seen some crates using "sysfs" for configuring PWM pins (for example, [rppal](https://crates.io/crates/rppal) and [sysfs-pwm](https://crates.io/crates/sysfs-pwm)). Should I start using sysfs for doing that?

fpagliughi, 2020-11-06:
The GPIO chardev system is separate from the PWM interface, so it's a matter of which crate is targeting which Linux API. The *sysfs-pwm* sounds like what you may be looking for, but it appears to have a single release from five years ago, so I suspect it could use some testing and probably an update.

posborne, 2021-09-20:
I'm going to close this, I am the original author of this crate and sysfs-pwm which has moved ownership to the wg, but it does definitely need a fresh review and a release.

See https://github.com/rust-embedded/rust-sysfs-pwm/issues/12

closed 2021-09-20 by posborne