rust-embedded/gpio-cdev #49: Does Line.request()

tal-zvon, 2021-02-25:


tal-zvon, 2021-02-25:
Does it matter what we set this arg for an input line? If not, can we get a short note in the docs that it doesn't matter. Maybe have it be an `Option<u8>` that can be set to `None`?

JonathonReinhart, 2023-12-18:
> Does it matter what we set this arg for an input line?

Nope. I was [in the neighborhood](https://github.com/rust-embedded/gpio-cdev/issues/78#issuecomment-1861562611) and tracked this down.

In the uAPI, `default_values` is [documented](https://github.com/torvalds/linux/blob/v5.15/include/uapi/linux/gpio.h#L395-L397):

> ```
>  * @default_values: if the %GPIOHANDLE_REQUEST_OUTPUT is set for a requested
>  * line, this specifies the default output value, should be 0 (low) or
>  * 1 (high), anything else than 0 or 1 will be interpreted as 1 (high)
> ```

So it only applies for `REQUEST_OUTPUT`.

This is confirmed by looking [in the kernel](https://github.com/torvalds/linux/blob/v5.15/drivers/gpio/gpiolib-cdev.c#L353-L357):
```c
		if (lflags & GPIOHANDLE_REQUEST_OUTPUT) {
			int val = !!handlereq.default_values[i];

			ret = gpiod_direction_output(desc, val);
			if (ret)
				goto out_free_lh;
		} else if (lflags & GPIOHANDLE_REQUEST_INPUT) {
			ret = gpiod_direction_input(desc);
			if (ret)
				goto out_free_lh;
		}
```

You can see that for `REQUEST_OUTPUT`, the value from `default_values[i]` (which is the `default` arg in Rust code) is passed to `gpiod_direction_output()` to actually set the mode to output and the default state.

For `REQUEST_INPUT` however, `default_values[i]` is not used. (It's not used anywhere else in that function.)

I think a short note in the doc to say it only applies for outputs would make sense. Changing it to `Option<u8>` would just add unnecessary complexity, IMO.