rust-embedded/gpio-cdev #51: Seeming deadlock on select!

datdenkikniet, 2021-03-06:
Hello,

When using the following snippet in my code (as part of a Wiegand reader):

```
        let monitor_0: Next<AsyncLineEventHandler> = events0.next();
        let monitor_1: Next<AsyncLineEventHandler> = events1.next();

        pin!(monitor_0, monitor_1);

        select! {
            res = monitor_0 => { data = (data << 1) | (value & 1) as u64; },
            res = monitor_1 => {data = (data << 1) | (value & 1) as u64; },
        }
```
it seems like the code ends up in a Deadlock in the select! macro after running this snippet for 128 times, after which the program seems to end up in a deadlock.

Introducing a millisecond delay at the very top increases the amount of bytes that can be read to 256.

Introducing a larger delay seems to remove the deadlock all together, but the ordering of the events is lost causing the data to become garbage as the order of the events determines the final data output.

I'm not certain if this is a gpio-cdev problem, a kernel problem, or if I'm simply doing this incorrectly.

Any feedback is highly appreciated.

ryankurte, 2021-03-06:
huh, interesting failure! i have a bit of async experience but, the internals of this are unfamiliar to me... @mgottschlag no obligations but, any ideas?

absolutely grasping at straws but:
- have you tried reading from two pins on different chips to check whether it could be an interaction from sharing chips?
- i would be interested to know whether the behaviour changes if you restrict the executor to a single thread (not sure this will catch anything but, just in case it was related to https://github.com/rust-embedded/gpio-cdev/issues/50 or similar to the classic https://github.com/rust-lang/rust/issues/71072#issuecomment-621301778) 
- have you tried to run it under `gdb`? it can be pretty rough where async is involved but will usually highlight the problem area (ime `strace` and `cargo flamegraph` can also be useful for getting a grasp of these things, though raspbian is missing a bunch of kernel tools so this may not be workable for you)

datdenkikniet, 2021-03-07:
Thanks for your response :)

I've tried running it with `strace`, but then I get the same behaviour as with a longer-than-a-millisecond delay where deadlock seems to disappear but the ordering gets all messed up (presumably because of the overhead that `strace` induces). I'll be sure to try it with GDB, and the other two experiments you've mentioned. 

A detail I forgot to mention: I'm running this in a Buildroot environment on kernel 4.19.97, in case anyone is trying to reproduce this issue. I've not quite figured out how to build a newer kernel this way, so I hope that this version isn't missing some important fixes to the gpio device driver.

datdenkikniet, 2021-03-07:
Additionally: I'd completely missed it while trying to look for similar issues, but it seems that I'm effectively trying to accomplish what https://github.com/rust-embedded/gpio-cdev/issues/18 is describing. Sadly no real solution to the problem/how to do it either, so this is just what I came up with.

datdenkikniet, 2021-03-08:
When running this with `gdb`, it seems like the main thread doesn't actually start/get to the `futex_wait` syscall that it goes to when in "working" condition.

datdenkikniet, 2021-03-09:
After some more investigation, it seems that it might be an issue in Tokio itself, somehow. I made my own fork and updated it to Tokio 1 (https://github.com/rust-embedded/gpio-cdev/compare/master...datdenkikniet:tokio-update), but it seems to exhibit the exact same behaviour. What happens is: the `poll_next` function is called many times, but seemingly never completes. In the background it must, since the memory usage of the program does not change, but somewhere something is not going right.

mgottschlag, 2021-03-09:
I am sorry I cannot contribute - I actually do not have much low-level tokio experience.

Anyways, so the summary right now is:
- The problem is not caused by incorrect usage of `select!`? I had problems with `select!` in the past - the macro has non-obvious failure modes, requirements to the futures which trigger those failure modes when not met, and I kept confusing the two different macris from `futures` and `tokio`, so it would be nice to rule out `select!` as a reason for the problems.
- The problem is not actually caused by waiting on multiple lines, as `await` on one line instead of `select!` on two does not fix the problem? This is weird, the code used to work fairly reliably for me for such use cases. I use `tokio::select!` with a single line without problems.

datdenkikniet, 2021-03-09:
Whoops, pressed the wrong button.

Thanks for the feedback.

Yes, your summary seems to reflect what I've found so far. AFAIK I'm using `select!` correctly, and I'm definitely using Tokio's `select!`.

I'll re-try and remove all of the references to the 2nd line and re-try, maybe it starts working again in that case.

closed 2021-03-09 by datdenkikniet

datdenkikniet, 2021-03-11:
So, I wasn't entirely correct:

What happens is that the `file.poll_read` (or its equivalent in the older version) is simply never ready. I'd missed that the `ready!` macro actually returns `Poll::Pending` if the result of the `Poll` that is passed as an argument is also `Poll::Pending`.

Very much unsure why it happens, but seems to be the same result in the older version.