rust-embedded/gpio-cdev #52: Please include Async documentation on docs.rs

tal-zvon, 2021-03-08:
There is no mention of `AsyncLineEventHandle` anywhere on docs.rs. The only way to see this documentation is to build it locally with `cargo doc`, which is a bit of a pain, especially since many of the people using `gpio-cdev` are working on a raspberry pi, and if they are running the lite version of Raspbian, without a GUI, as I am, there is no browser, so `cargo doc --open` won't work. We need to manually go in and copy the docs directory to a different computer to read it.

When I did `cargo doc` on my project, it generated 16k files. Using scp to copy them over to my other computer was going to take like half an hour. I had to interrupt the scp copy operation, tar the 16k of doc html files, transfer it as one file, and unarchive it before opening it. The whole process took like 20 minutes. It would have been nice if I could have just found the documentation on docs.rs in seconds, as I do with most other documentation.

nelsonjchen, 2021-03-08:
Someone asked about this on SO:

https://stackoverflow.com/questions/66519446/is-it-possible-to-see-feature-documentation-on-docs-rs/66519709#66519709

nelsonjchen, 2021-03-08:
@tal-zvon Not great, but you might be able to use something like `python3 -m http.server` to just serve the doc html directory and browse the files from your workstation. Should be faster than SCP/tar-ball/etc.

tal-zvon, 2021-03-08:
Yup - that's one way to go. Would be nice if we didn't need to jump through hoops to see that section of documentation though.

ryankurte, 2021-03-08:
Hey there, thanks for opening the issue. It's possible to use docsrs attributes (`#[cfg_attr(docsrs, doc(cfg(feature = "async-tokio")))]`) to show these features on docs.rs, which were recently added here, (hopefully) just need to publish a release.

tal-zvon, 2021-03-08:
Awesome! I'd close the issue, but it seems that it's set to auto-close when 0.4.1 is released. I guess I'll keep it open for people who run into the same issue, so they can find the answer easier, until the 0.4.1 is released, which will both solve, and auto-close this.

Thanks

commit bd58cf1a (2021-09-21): Merge #64

64: Release 0.5.0 version r=ryankurte a=eldruin

Fixes #52
PR #64 (2021-09-22): Release 0.5.0 version
Fixes #52
closed 2021-09-22 by bors[bot]